makerbase-mks/MKS-TFT #437: TFT32 V3.0 some Icons displaying poorly

bartofborg, 2020-08-04:
updated to 3.06 release and connected to SKR v1.3. board. some icons are great others you can barely see. not sure what is going on. I plugged it into the serial port for the board before updating and the icons where all great but none of the functions worked. I updated the display and now the buttons work but the button icons are poorly displayed. any idea how to fix this?
[image]

makerbase-mks, 2020-08-11:
Hi! You need redowanload firmware and mks_pic floder,reupdate it.

bartofborg, 2020-08-24:
tried that twice, used retro and classic - still the same issue

bilou24an199, 2021-02-07:
hello for my part,

for the tft35 i had the same probleme and to remove the poor icons, i flash the TFT with an old firmware (1.0.2 the actual one is 1.0.7).

flash in 2 steps =
 -  firmware - font - config.txt
 - then wifi + pic (with std thme)
 
and in second step = apply of new theme 

and on the last step = add the new firmware