makerbase-mks/MKS-TFT #438: MKS TFT 35 90 degrees Screen Orientation

Cemileri, 2020-08-06:
Hello everyone, I was wondering if there is a way to use this screen 90 degrees vertical with all the buttons and text oriented in this way. Thanks in advance..

makerbase-mks, 2020-08-11:
Hi！ It is looks greatful, more users need to use it like this?

Cemileri, 2020-08-18:
I think so..

Cemileri, 2020-09-10:
> Hi！ It is looks greatful, more users need to use it like this?

Please add this functionality or show me a way to do it. Thx

mshavandi, 2020-09-28:
I have this problem too. Did you solve it?

selvakumaran2016, 2020-10-05:
Yes, this will be useful for some machine makers, 90 deg rotation in both clockwise and anticlockwise direction

mshavandi, 2020-10-06:
@selvakumaran2016 Yes, exactly, my design needs this orientation. Did you find a way to implement it? I'm struggling for a week but no luck.

selvakumaran2016, 2020-10-07:
@mshavandi Sorry, even I need that 90 deg orientation.

tomasbaldi, 2021-03-01:
Yes, exactly, my design needs this orientation!!