makerbase-mks/MKS-TFT #439: MKS TFT35 stop comunicating

fr4231gm, 2020-08-08:
The touchscreen stops comunicating randomly to my Robin e3d board, that makes stop prints at mid or stop send orders to de printer. Software version 1.0.7.

fr4231gm, 2020-08-10:
Please I need help :(

makerbase-mks, 2020-08-11:
Hi! What is your settings baudrate? If use 250000, you can try to setting to 115200.

fr4231gm, 2020-08-13:
Ok thank you! But now I have other problem, I have calibrated the touch screen and now doesnt work, maybe because I didnt calibrate it in the good way, how can I reset that calibration?

fr4231gm, 2020-08-13:
Solved. Thanks

makerbase-mks, 2020-08-14:
Can you share your problem solving process？

fr4231gm, 2020-08-18:
Hi! I have done some long prints, about 10 hours with the MKS and it worked perfectly. The problem of the calibration was solved re-updating the firmware of the TFT, and the problem of the connection, IDK how, but, I connected my other screen to the printer too because I couldnt work with the MKS TFT and I connected too the SD Card in the Robin e3, and when I connected both things, the MKS started to work perfectly, I know that this has no sense, but magically worked for me xD. Thanks for everything!

fr4231gm, 2020-08-18:
And I have to say that I didnt change the baudrate. It still using 250000

makerbase-mks, 2020-09-03:
@fr4231gm 
OK, then I will close this question temporarily.

closed 2020-09-03 by makerbase-mks

acca84, 2020-11-17:
@fr4231gm, i think i have the same problem with MKS SGEN L V1.0 :  https://github.com/makerbase-mks/SGEN_L/issues/46
Do you have a dual hotend setup ? because i can print with one hotend but not two, print stops without reason.
did you find a workaround for using only the TFT35 ?
thanks