makerbase-mks/MKS-TFT #440: Tft 35; print not strarting

mcsultan, 2020-08-08:
i had 1.0.3 firmware. few days ago everytink was ok. Now, when i confirm the file for printing, status page is opening but print not starting. Nozzle and heatbed not going warm. only the counter works.

try printing after  use preheat also same, but if i click pause button and resume button, print is starting normaly.

I update firmware to 1.0.7, but no change.

mks 35+ mks gen_L, marlin 1.1.8

Please help.

Error26, 2020-08-09:
This is a firmware problem 1.7.0. I also have problems with printing on Marlin and Repetier.

makerbase-mks, 2020-08-11:
Hi! I want to you add picture and it operation.

mcsultan, 2020-08-11:
ok, ill try explain and sending pictures
Im using simplyfy3d for slicing, and using mks_host for picture code. Also im using window 10.
i create 2 files. one of for gcode(simplyfy3d) one of picture code(mks_host).
i copy picture code from mks_host file and paste to simplyfy3d file then i have one windows (CRLF) type file. Or;
i copy simplyfy3d  gcode and paste to mks_host file then i have one Macintosh(CR) type file.

Before both file was working nicely.
But now; if i create Macintosh(CR) type file with image code, item of image looking good but printing not starting. just counter is working. Heatbed or nozzle not going to warm.

if i create windows (CRLF) type file with image code; the image looking corrupted, but print process is starting.
if i use sipmplyfy3d file directly (windows (CRLF) type file without image code) also print process is starting.

[image]
[image]
[image]
[image]
[image]
[image]
[files.zip](https://github.com/makerbase-mks/MKS-TFT/files/5059391/files.zip)

Gorkag3d, 2020-09-10:
Hi,
I have the same issue, MKS Robin Nano and Tft35. Using Mks_Nano35_V1.0.3 Release file.
I use Cura with mks plugin
After printing properly, the next attempt to print the bed and the nozzle dont warm up. I have tried all of them and works properly.
I had also been able to print using wifi from my mobile, but not directly from the Tft screen.
The only thing different I have done and I will try today is to delete a non GCODE file from the SDCARD before slicing, I guess is the only thing I had made different.
At least I had managed to print using wifi.
Best 3d wishes,

BMKMotorentechnik, 2021-09-23:
Will try the same today - having the same Issue on my compiled 1.0.8 - strange thing : even when its just Gcode.
Trying to find some hidden data on the card or somewhat in the evening.

What works until yet : Aborting the print and restarting it.