makerbase-mks/MKS-TFT #443: MKS TFT 3.5 firmware is bad after cancel a print job

daniellezama, 2020-08-17:
Hi I need to restart the printer after cancel a job because when i send the print again not make the heating of bed and hotend and start to print to 50% of program and after  continue printing without heat the bed and the noozle. 

i am using latest firmware update.

Regards

Daniel Leza,a

daniellezama, 2020-08-18:
The printer when starts at once all ok. First Job ends correctly, wifi sended. When ends, and try to send a second job or print other job from sd, start running in the air without g28 home and no heating, must turn off the printer, then come back ok. Every time must restart the printer for make work it.

daniellezama, 2020-08-19:
i have retested today with lcd12864 display from SD card marlin runs great all ok. No problems. Then i have retested the TFT and the problem remains. Only one print job sucessful. Second job runs wihout warming without extrusion and without homing. I am sure that is a firmware bug. Comunication is fine un 57600 bauds

daniellezama, 2020-08-20:
Well after trying the other versions of firmwares could conclude version 1.05 stable but no gcode window and no tft image when printing) ,
version 1.06 stable with gcode window and tft image when printing, but when send the program remotely not show the print image. 
version  1.07 unstable show the print image and run fine only first printing. next prints fly in the air. Also problems for detect de sd ( must shut down and restart for read the SD)

jechs83, 2020-09-01:
Same problem here with 1.07, it looks  really nice with more configurations functions but its impossible to print normally after first print

CBDesignS, 2020-09-01:
yes I have the same with 1.0.7 as above. it seems to just continue from where it was stopped without re leveling or even heating or extrusion. . I think tft pretends to be a computer controlling main board by serial data but the serial data is not reset so the tft just starts from the last command without realising it has been stopped, cooled down etc. bad firmware coding or config. please sort this out asap.

daniellezama, 2020-09-01:
Yes i have the same configuration CBDesignS and must choose the 1.0.6 version more stable and works. Only problem is when send the tft gcode from computer  not show the tft image when print. 

Regards

Daniel

CBDesignS, 2020-09-01:
daniellezama I had a simmilar problem with the tft gcode preview messing up untill I updated cura to 4.7.0 and manually updated the tft/wifi plugin from the github code page and now the preview is ok.
[image]

before it was random lines and only part of the preview

jechs83, 2020-12-05:
Still waiting tft 35 firmware 1.0.8 and makerbase doesnt care about it...  I think we should try other brand options by now.

CBDesignS, 2020-12-05:
I removed the mks hardware and jumped to btt because at least the userbase  is willing to support the hardware better than bigtree is doing and MKS ever will.  The TFT has just as many problems but at least users are skilled enough to fix the problems that have arrise from time to time