makerbase-mks/MKS-TFT #444: TFT24 frozen on babystepping screen

McPaul0, 2020-08-19:
If I press babystepping button during first layer printing it stops.
If I Press babystepping button before print, it does not start to print at all.
When you Fix this problem?

makerbase-mks, 2020-09-18:
MKS TFT24 firmware is latest version?

McPaul0, 2020-09-18:
> MKS TFT24 firmware is latest version?
firmware  V_3.0.6
but V_3.0.5 is the same

selvakumaran2016, 2020-10-05:
Even I am facing the same problem with MKS TFT35. When I press babystepping it is freezing. When I asked someone from Makerbase, they told that babystepping will not work with Marlin 2.0.6

I hope this is a must required one.

McPaul0, 2020-10-05:
But it works with additional buttons. I've edited config file and it works
 #command setting on button1~6
>morefunc1_cmd:M290 Z-0.025;
>morefunc2_cmd:M290 Z0.025;

selvakumaran2016, 2020-10-20:
@McPaul0 Even we tried the above commands. Still it is freezing.

@makerbase-mks 
In Skype, I raised a query. I used RUMBA 8 bit board with Marlin 2.0. In that, I used TFT 3.5". When we used baby-stepping, the screen freezes. But you told that it 8-bit board is not compatible with the Latest TFT firmware. So, baby-stepping will not work.

But, now I am using the S-Gen L 32-Bit board. I am using Marlin 2.0 in which Babystepping is enabled.
I am using Latest MKS TFT 2.4" firmware. I have enabled Babystepping in TFT firmware also. When printing, if I touch the baby-stepping button, the screen is freezing. 

Can you please check at your end and update the Firmware for us? Babystepping is really important. 

Also, we are expecting some support from you. We are feeling that the support from Makerbase is really slow and you are not giving importance for customer queries.

In this case, we have to switch to the other boards.

selvakumaran2016, 2020-12-03:
@MKS-Sean @mks-viva @makerbase-mks Can you please look into this issue? Even I have send the same to Mr. Cai through Skype. Will appreciate if you are providing the help.

mks-viva, 2020-12-04:
@selvakumaran2016 
Hi! Thank you for your feedback, we will test it and reproduce the problem.

Demonora, 2020-12-26:
Hi , i am using the mks S Gen L Board and the tft 2.4 v1 screen and while i print it stops then print again after 30 second 
and it repeats over and over . Please help

mks-viva, 2020-12-26:
@Demonora 
Hi! Can you provide your firmware to me and i need test it

Demonora, 2020-12-26:
@

Demonora, 2020-12-26:
Where do i send it?

mks-viva, 2020-12-29:
Are you using the latest version of marlin firmware? And the firmware of MKS TFT24, has it been updated to the latest version? If not, then I recommend updating to the latest version to test it

Demonora, 2020-12-29:
Marlin and the mks tft is up to date and the mks tft are freezing now when
i put in sd card. Then i must do a hard reset

mks-viva, 2020-12-30:
> Marlin and the mks tft is up to date and the mks tft are freezing now when i put in sd card. Then i must do a hard reset
> […](#)

Have you tried to test with other cards?

Demonora, 2020-12-30:
The only other card I have are the ramps 1.4 and it is not compatible with
the mks tft . Cards are very expensive where I live ,this was the best card
and screen in my budget. .