makerbase-mks/MKS-TFT #449: tft 35 problems

AlexejLach, 2020-09-05:
wi-fi does not work with firmware 1.0.6 what to do?

makerbase-mks, 2020-09-10:
Can you provide your config to me?

AlexejLach, 2020-09-10:
mks sgenl v2 - mks tft35 - mks tft wifi 

with firmware 1.0.7 everything worked

AlexejLach, 2020-09-10:
#--------------------------------------------------------------------
#########  Printer type setting  ##################################

#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:1

#machine setting (Normal:1; Delta:2)
>cfg_machine_type:1

#baud rate (9600:1; 57600:2; 115200:3; 250000:4)
>cfg_baud_rate:4

#multi-language(enable:1, disable:0)
>cfg_multiple_language:1

#languages setting(simplified (simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).
#This configuration is valid when "cfg_multiple_language" is disabled.
>cfg_language_type:3

#extruder number(one:1; dual:2)
>cfg_sprayer_counter:1

#enable heated bed(yes:1; no: 0)
>cfg_custom_bed_flag:1

#the max target temp of extruder and heated bed
>cfg_max_sprayer_temperature:250
>cfg_max_hotbed_temperature:120

#pause position (-1 is invalid; Z-axis is relative position)
>cfg_XPOS:-1
>cfg_YPOS:-1
>cfg_ZADD:10

#-----------------------------------------------------------------------
#########  Advanced Function Option  ##################################

#enable UPS?(yes:1; no:0)
>cfg_have_ups_device:0

#enable power detecting module(mks 220det:1; mks pwc:0)
>cfg_insert_det_module:0

#enable auto off after print finish function (no:0; Yes:1）
>cfg_print_finish_close_Machine:1

#set PB0 signal when use it for sencond nozzle (high level:1; low level:0)
>cfg_PB0_trigger_Level:0

#set PB1 signal (high level:1; low level:0)
>cfg_PB1_trigger_Level:0

#set error range of Z-axis on breakpoints recovery 
>cfg_breakpoint_z_error:0.2

#the function of power detecting and filament detecting(1:disable;0:enable)
>cfg_mask_PB0_PB1_function:1

# the method of save the gcode data with power off(1:Capacitor storage; 0:File system save)
>cfg_pwroff_save_mode:0

#---------------------------------------------------------------------------

#########  Filament Change Function  ##############################

#the speed to extrude filament(mm/min)
>cfg_filament_load_speed:1200
#the lenght to extrude filament (mm)
>cfg_filament_load_length:200

[... 162 lines cut ...]
#the color of dialog button background
>cfg_dialog_btn_bkcolor:0x000000

#the color of dialog button text
>cfg_dialog_btn_textcolor:0xFFFFFF

#3D effects setting (disable: 0; enable: 1)
>cfg_BUTTON_3D_effects:0

#position of button font offset bottom(unit:pixel)
>cfg_btn_text_offset:10

##Screen rotation 180 °(0:disable,0xEE:enable)
>cfg_screen_overturn_180:0x00

#the style of display on the screen home(0:classic,1:simple)
>cfg_screen_display_mode:1

#Enable function of display error message. (no:0; Yes:1）
>cfg_display_eeror_message:0

#Enable function of babastep. (no:0; Yes:1）
>cfg_babystep_btn_display:1

#-----------------------------------------------------------------------

makerbase-mks, 2020-09-10:
Do you want to connect WIFI to your network? If yes, you need to set to：
############# WIFI Function 
#wifi mode(AP:1; STA:0)
cfg_wifi_mode:0
#wifi name and password
cfg_wifi_ap_name: your network
cfg_wifi_key_code: your network password

#0:disable cloud service 1:enable cloud service
cfg_cloud_enable:1

#cloud service adress
cfg_wifi_cloud_host:www.baizhongyun.cn

#cloud service port
cfg_cloud_port:10086

#Dynamic access IP
cfg_ip_dhcp_flag:1

AlexejLach, 2020-09-10:
thank you everything worked :)

makerbase-mks, 2020-09-10:
OK! I will close it.

closed 2020-09-10 by makerbase-mks