makerbase-mks/MKS-TFT #450: TFT32: Update

SkipperDo, 2020-09-08:
Its just a minimalistic issue...
Everytime i update the displays firmware, every time it shows "V3.0.0" no matter what version i install.. 
It should be better to display the Firmware Version i install currently to avoid confusion..

makerbase-mks, 2020-09-09:
Ok! thank your feedback.

BenediktS, 2020-11-22:
+1 : I was just confused about the same thing

MKS-Sean, 2021-02-04:
@SkipperDo @BenediktS Hi，In fact, you can check your version in the about screen, this V3.0.0 is just the version of the bootloader

SkipperDo, 2021-02-04:
@MKS-Sean : OK... I Understand... But is is a big thing to write in the screen "Bootloader: V3.0.0" instead of just "V3.0.0"?
Maybe this screen can show the Firmware Version too? Like:
Bootloader: V3.0.0
Firmware: V1.2.3.4

for an example?
Would be great...