makerbase-mks/MKS-TFT #452: MKS TFT35 not communicate with MKS DLC in laser mode grbl

ksdfpr, 2020-09-09:
Hello, 
My problem is that FT35 not communicate with MKS DLC in laser mode grbl. I have changed firmare in DLC already and in TFT also (v1.0.2) also. Both works fine but without any communication among them. I have check the baudrate to 3 in config file (115200) according to DLC baud rate but nothing. Any idea?

makerbase-mks, 2020-09-10:
Use laser TFT firmware?

ksdfpr, 2020-09-10:
Yes, the file MKS_TFT35_laser_V1.0.2 Release file.zip

makerbase-mks, 2020-09-10:
I think your GRBL has lock, you need send $X unlock it, you can use pronterface connect DLC motherboard check it.

ksdfpr, 2020-09-10:
I checked it. No reaction

ksdfpr, 2020-09-12:
... and while I have installed this  file MKS_TFT35_laser_V1.0.2 Release file.zip, in the About menu says "MKS TFT_35_LASER VERSION 1.0.1 (and please change the word "conected" with "connected")

ksdfpr, 2020-09-14:
Any other suggestion?

ksdfpr, 2020-09-19:
FINALLY!
 The problem: is the auto lock function is enable when grbl starts up. If you have a pc connected, its not a problem, just send a $X command and after that the communication with TFT works. But if you dont have a pc connected to send this command, the communication betweeen DLC and TFT is not working.
The answer: in the config.h  file in grbl you can comment HOMMING_INIT_LOCK  or autohoming on boot, (not the best solutions both...)

makerbase-mks, 2020-09-19:
So, MKS TFT35 Laser needs to add the unlock command button.

ksdfpr, 2020-09-19:
Yes, I have checked it. If you press some buttons, the communication stops for some seconds, so it must be on the first screen.
There is missing a home button also (to reach limits, not hom to origin)
Also please correct the "conect" with "connect"
Thank you

bouba830, 2020-11-11:
avez vous réussi car je me retrouve avec le même problème

mks-viva, 2020-12-30:
@ksdfpr The latest version has fix it. So, i will close this issues temporarily.

closed 2020-12-30 by mks-viva