makerbase-mks/MKS-TFT #455: MKS-TFT + MKS-WIFI incompatibility

scmanjarrez, 2020-09-13:
Hi,
Recently I bought the mks-wifi module, but if I try to print using uploaded to the SD card files from the tft or print from mks wifi plug-in for cura, prints start doing wrong things like stopping mid-print.
Any fix?
Are you planning to release mks-wifi source code?

kekko-maker, 2020-10-07:
Afther you upload the file .gcode, then insert the USB or SD card and try to check if there Is some strange code before the actual gcode, this Is probably linked to the cura plug in and Is the screen shot of the 3d file. So t
Try ti delete It ad re run the print , if It works Is a problem linked to cura plug in and all you Need to do Is to remove the functions that take the screenshot from settings. Please check if It Is this the problem and i Will explain you how to solve It.

scmanjarrez, 2020-10-07:
Hi, thanks for your help.
This only happens when wifi is enabled, if I disable it, stops don't happen. I'll check again for screenshot gcode. Thanks