makerbase-mks/MKS-TFT #462: SGEN-L /  TFT35 filament sensor problem

hardy8, 2020-10-04:
Hello 

I'm trying to connect filament runout sensor on MKS GEN-L board with TFT-35 touch screen. I made necessary changes to use filament sensor on board connection but my TFT-35 screen doesn't seem to like that very much.
While filament sensor connected on board, when i pull out the filament print is stopped and nozzle parked the determined position as it should be, but TFT-35 doesnt react this move and acts like print still continues. Sad thing is while filament is pulled out and nozzle is parked, i tried to stop/pause on TFT screen but nothing happens. Seems like runout sensor and TFT not communication while sensor connected on board.
TFT has its own pins allocated for filament sensor but i need to customise printhead moves for my printer, so TFT pins are not an option for me.

Do i necessarly use TFT pins or can i find a way to make TFT talk to Marlin.

Hardy

makerbase-mks, 2020-10-06:
You need filament runout sensor signel connect to MKS TFT35

mks-viva, 2021-01-06:
Hi! I will close this issues temporarily.

closed 2021-01-06 by mks-viva