makerbase-mks/MKS-TFT #466: Blower Fan to 100% with AUX1 display connected

panto91, 2020-10-08:
Hello,
i installed a TFT35 with WIFI to my MKS GEN L V.10 board via AUX-1 port.
There are some issue with the fan.
When the display is connected to the board and i switch on the printer, the blower fan starts blowing at 100%.
If i go to settings, fan, i see that the fan is off for the firmware and also for the display.
If i tap on 0% or on 100% or other button in fan setting the display become total WHITE and i can't use it unless i detach the plug or i switch off the printer.
I tried also to change the speed with the stock display RepRap but the TFT turns white anyway.
If i don't connect the TFT35 all works fine and the blower fan is off on startup.

Please help me!

Regards

mks-viva, 2020-10-15:
If you only use MKS TFT35 and it's will work?

panto91, 2020-10-15:
> If you only use MKS TFT35 and it's will work?

With only MKS TFT35 NOT work!!!
With only reprap works perfectly...
With rep rap + TFT35 the blower is on but on the two displays the blower seams off (0).
If I try to change that value to 255 or to another value ... the firmware crash! 
Any help?

panto91, 2020-12-24:
Anyone can help me?

mks-viva, 2020-12-25:
Is MKS TFT35 the latest firmware? Can you provide vodies when you use MKS TFT35 connect MKS Gen_L V1.0?

panto91, 2020-12-25:
Here the video 

https://user-images.githubusercontent.com/25211647/103130320-7cf1dc80-469c-11eb-9fca-4cfbc6940a57.MOV

as you can see in the video if I try to stop or start the blower it is continue to blow at 100%.
When I use + o - the firmware crash and the display become white

mks-viva, 2020-12-26:
Which interface are you FAN connecting?

panto91, 2020-12-26:
> Which interface are you FAN connecting?

To the right pins of the blower so D7 pins

mks-viva, 2020-12-30:
Is FAN turned on by default when it is powered on?

panto91, 2020-12-30:
> Is FAN turned on by default when it is powered on?

Yes but it is a news ... some weeks ago was not! When I power up the printer ... the fan must be off!

mks-viva, 2020-12-30:
But this is the opposite of most users.

panto91, 2020-12-31:
> But this is the opposite of most users.

When start a print the blower must be OFF for first layers. If I put a Gcode the switch off the blower, this blower continue to blow at 80%... this is not standard.
Now I tried to change board, to a new mks gen l v2.1. I hope this helps.