makerbase-mks/MKS-TFT #468: PROBLEM WITH MKS GEN L V2.1 WITH MKS TFT35 wifi

onionboy69, 2020-10-17:
Hello, I managed to put up this sistem on a Tevo Tarantula with the latest firmware on the github...

 I have some problems at printing, becouse it stops when I click on the Gcode file, with a Continue Button that does  nothing, also the calibration button doesn't becouse when I enter the calib mode just freeze and don't stop the calibration process...
PLEASE HELP ME, I PUT YOU HERE WHAT CONFIG I USE.
THANK YOU

[image]

[image]
mks gen l v2.1 marlin that i used
[Marlin.zip](https://github.com/makerbase-mks/MKS-TFT/files/5395858/Marlin.zip)
tft35 firmware and setting used
[TFT35_V1.0.7 Release file.zip](https://github.com/makerbase-mks/MKS-TFT/files/5395866/TFT35_V1.0.7.Release.file.zip)

onionboy69, 2020-10-17:
first photo of the tft is when I printed over usb, print blocked after few min of print...
if I click confirm after some seconds a new error with a new line appear.

selvakumaran2016, 2020-10-20:
@onionboy69 I also faced the same 'Confirm' issue when I updated the latest firmware for TFT 35. I contact MKS, they updated the mkstft35.bin and send it to me.

you can unzip this file and update in TFT. It will solve the problem of 'Confirm' issues

[mkstft35.zip](https://github.com/makerbase-mks/MKS-TFT/files/5408225/mkstft35.zip)

onionboy69, 2020-10-20:
hello, I updated the TFT, same problem, I think I might have someting in the marlin worng.
this problem gives me a headache, I tried various methods, only one time the tft worked and the nozzle and bed didn't read temp. but like a bigginer :)) I forgot what thing in the code I changed... 
this is usless.. 
anyways thannks for the suport @selvakumaran2016 
any help someone?

selvakumaran2016, 2020-10-22:
@onionboy69 Have you coped the above mkstft35.zip into the TFT firmware folder? and replace the existing file?

Delete all the contents from SD card. Copy the TFT Firmware. Insert into TFT and switch on the machine.

It will install the new firmware. The above 'Confirm' problem will be solved

onionboy69, 2020-10-22:
@selvakumaran2016 Allready done that..

onionboy69, 2020-11-04:
Help?
@makerbase-mks

onionboy69, 2020-11-08:
yoooooooooooooooooo
anyone here?