makerbase-mks/MKS-TFT #471: MKS SGENl/TFT35 babystepping

hardy8, 2020-10-29:
I use marlin 2.0.5.3 and all works fine, but now i made some changes and also switched to marlin 2.0.7.2. but now baby stepping doesn't work anymore, (the display freezes).
Why is this and how do i get it back working again, or is this a bug, so do I have to wait for an update?

Greetings Hardy

mks-viva, 2020-11-06:
Hi! I suggest to use 2.0.5.3version temporarily.

mks-viva, 2020-11-20:
I will close this issues temporarily

closed 2020-11-20 by mks-viva

hardy8, 2020-11-20:
Too bad, I had hoped that I could also use the latest versions of marlin, along with the improvements.

mks-viva, 2020-11-23:
I am still very worried. There will be many bugs in the newly released version of marlin. For example, the function of POWER_LOSS_RECOVERY failure in 2.0.7.2 is more imperfect than 2.0.6...So, if 2.0.5 can be used stably, wait until 2.0.7.2 is resolved And then update it.

hardy8, 2020-11-23:
Thanks ! I will wait.Regard HardyVerzonden vanaf mijn Galaxy

mks-viva, 2020-11-24:
We look forward to the new version of the firmware with more complete functions.

V3nn3tj3, 2021-02-05:
Problem still exists with Marlin 2.0.7.2 and SGEN L V2. Any idea when this will be solved?