makerbase-mks/MKS-TFT #474: Problème de communications entre MKS SGEN L V2 et KMS TFT70 V1.1

TRANIER, 2020-11-14:
Bonjour,
Mon MKS-TFT70 est bien allumé , carte wifi fonctionne, mais, aucune commandes est exploitable, pourtant la câble et les connections sont bien vérifiés
1/ quelles sont les paramètre dans "configuration.h" a modifier ?
2/ firmware à installer ? je suis dans le mur !!

Merci pour votre aide et vos idées.

[image]

mks-viva, 2020-12-30:
Hi! I want to know. Can you check wifi by MKS TFT70 display screen?

tschork, 2021-02-03:
See English below, if it can help someone.

@TRANIER si ça peut aider, j'ai du recompiler mon firmware pour que le tft35 fonctionne avec une mks Sbase v1.3 sous Marlin 2.

les 3 choses que j'ai du modifier sont:
```
//pour communiquer en série sur le port AUX1
#define SERIAL_PORT 0 

//désactiver le support sd sur la carte mère
//#define SDSUPPORT

//désactiver l'écran configuré avant
//#define REPRAP_DISCOUNT_SMART_CONTROLLER
```
Une fois ces changements fais, j'ai pu avoir une communication entre le tft35 et la sbase.

-----------------

There are 3 things that I had to change in my Marlin 2.0 firmware to connect the TFT35 to my MKS Sbase v1.3.

```
//Activate serial on the AUX1 port
#define SERIAL_PORT 0 

//Remove main board sd-card support
//#define SDSUPPORT

//Remove config for a screen directly attached to the main board that I had before
//#define REPRAP_DISCOUNT_SMART_CONTROLLER
```

Once I did those changes and updated the firmware of my Sbase, then the TFT35 controller was able to communicate with the mainboard