makerbase-mks/MKS-TFT #475: [theme / skin] TFT 35

bilou24an199, 2020-11-15:
hello,

i want to change the std skin of my tft 35.
i have downloaded the application MKSTOOL which is a great solution to update the theme or to make a custom theme

in the standard position of your github

https://github.com/makerbase-mks/MKS-TOOL

a new theme is proposed = mks_picdiy

this theme is gorgeous, but there is only one pb = in the menu "HOME" =

[image]

the icon is not appearing for the "HOME" button is it a software bug or is this button icon is missing in the distribution ?

please can you provide th ICON to have a complet theme

Best regards

bilou24an199, 2020-11-15:
Ok i compare 2 folders = the folder of the std theme and the folder of diy theme
the button icon which is missing call =
bmp_zeroA.bin

for a first solution, i duplicate the bmp_zero.bin and i rename it = bmp_zeroA.bin

 it is may be not the good icon, but now, no icon is missing.

is it possible to have the photoshop files, because i would like and custom button (more button) and to match with the theme (blue button with black bckground and colored dots) it will be usefull to have the original files

thank in advance

best regards

bilou24an199, 2020-11-15:
some other std icons are missing =

the icon for file system =
- bmp_usb.bin
- bmp_usb_sel.bin

and may be more,

please put the archive from this skin (in illustrator or photoshop or other native files) in the TFT design UI

BR

mks-viva, 2020-11-17:
Thank you for your feedback, I will report this issue with our engineer.

mks-viva, 2020-12-30:
You can try to use it config: https://www.baizhongyun.cn/home/mkstoolview

bilou24an199, 2021-02-07:
hello i have not use your link but thank you for your help

i send you a zip file on email _**4164049@qq.com**_

with my "adaptation" of the mks_picdiy

i want to deposite it on the github but no access to do that or not found the mean to do that

BR

closed 2021-02-25 by mks-viva