makerbase-mks/MKS-TFT #478: Mks tft 32 cnc update error

ascarons22, 2020-11-17:
tft 32 1.02 update file gives sd error in 82%, tested with several cards, a solution?

mks-viva, 2020-11-18:
Hi! Can you provide a video for me?

ascarons22, 2020-11-18:
[
[mks.zip](https://github.com/makerbase-mks/MKS-TFT/files/5561878/mks.zip)

](url)

ascarons22, 2020-11-18:
Inside the zip is the video, it always hangs at the same point, tested with several cards.

mks-viva, 2020-11-20:
Is the SD card formatted?

ascarons22, 2020-11-20:
yes, formatted in fat32, with the 3d printer file it works, with version 1.0 it works, with version 1.02 always error in 82%.

mks-viva, 2020-11-21:
I will communicate this issue with our engineer.

kapsko, 2020-12-12:
im having same thing happen any update

secondstar1, 2021-01-10:
> I will communicate this issue with our engineer.
Any feedback from engineer? I have the same issue.

MKS-Sean, 2021-02-04:
@ascarons22 Hi,I think I solved this problem, just update to the latest firmware
#498

closed 2021-03-05 by mks-viva