makerbase-mks/MKS-TFT #480: mks tft 35 stopped seeing the board

AlexejLach, 2020-11-19:
mks tft 35 stopped seeing the board. everything was fine yesterday, but today he does not see her. I flashed the screen and checked the cable, nothing helps

AlexejLach, 2020-11-19:
https://github.com/makerbase-mks/MKS-TFT/issues/218

onsc, 2020-11-20:
Same problem here. I think uart is broken.

mks-viva, 2020-11-21:
Most likely, the problem lies in the serial port circuit

onsc, 2020-11-21:
> Most likely, the problem lies in the serial port circuit
Should we replace chip or wait to another firmware ? What do you think?

mks-viva, 2020-11-23:
It is difficult for me to find out exactly where the problem lies, because I don’t have similar hardware, but you can refer to MKS TFT35 hardware.
https://github.com/makerbase-mks/MKS-TFT-Hardware/blob/master/MKS%20TFT35/MKS%20TFT35%20V1.0_003/MKS%20TFT35%20V1.0_003%20SCH.pdf

onsc, 2020-11-23:
serial ports connected to directly stm chip. So is it possible only that pins are broken?
i connected to Arduino mega. And used multi serial. Same baudrate. And clicked some printer command buttons. Nothing shows.at Arduino side. is it good method to test stm serial? Thank you.

mks-viva, 2020-11-23:
This is serial hardware on mks tft35
STM32 PB11 RXD connect to AUX-1 RXD
STM32 PB10 TXD-1 connect to D9(Level conversion chip) , D9 TXD connect to AUX-TXD
[image]

onsc, 2020-11-23:
> Level conversion chip

Thank you very much for your interest. I just realized that level conversion chip. What do  you suggest me to check or any check method to analyze ? 
Maybe problem is the conversion chip ...

mks-viva, 2020-11-23:
D9 is : SN74LVC1T45DCKR
I think there is a high probability that D9 is not working properly, which leads to communication failure.

onsc, 2020-11-23:
> D9 is : SN74LVC1T45DCKR
> If firmware settings is OK. I think there is a high probability that D9 is not working properly, which leads to communication failure.

Thank you mks-viva. I ll try to replace in a week. I think firmware has right configuration and OK.

mks-viva, 2020-11-23:
OK! Good luck for you.

mks-viva, 2021-01-06:
Hi! I will close this issues temporarily.

closed 2021-01-06 by mks-viva

onsc, 2021-04-28:
Sorry for big delay. i couldn't get that part in that form. So i couldn't tried. Thank you.