makerbase-mks/MKS-TFT #483: TFT32 - After update to V3.0.6 no Heat Bed is showing up

BenediktS, 2020-11-22:
I just updated my "MKS TFT32_L V3.0" to the firmware version v3.0.6 . 
After the update the heating bed is not showing up anymore. 

It is not showing up in the pre heat screen nor in the printing screen.
In my configuration file this are the hotbed lines:

>cfg_custom_bed_flag: 1     # Please choose  1:Enable bed  0:Disable bed
>cfg_max_hotbed_temperature: 150     # The max target temp of heated bed
 
Did i miss something?

mks-viva, 2020-11-23:
Hi! You can check your baudrate settings in your config file.

BenediktS, 2020-11-25:
I checked it. 
It is the same baudrate  that is configured on the marlin board.

I could see the sprayer temperature and can use it. 
I can also print without problems. 
I could only not see the hotbed.  

Is the sprayer temperature using a different communication way then the hotbed?

mks-viva, 2020-11-26:
Marlin firmware have config hotbed?

BenediktS, 2020-12-12:
I have to update the firmware i will get back to you. I think i can test it next weekend.

mks-viva, 2020-12-14:
Waiting for your good news

mks-viva, 2021-01-06:
Hi! I will close this issues temporarily.

closed 2021-01-06 by mks-viva