makerbase-mks/MKS-TFT #487: MKS TFT 32 v3.0 + MKS GEN L 1.0 Black screen after inserting SD card

Anima1000000, 2020-11-28:
After having finished a print, taking out the SD card, uploading a new file and reinserting it and going to the print dialog, the screen turns black. This doesn't always happen, but about 50% of the time. Switching the printer off and on is fixing it no need to reinsert SD. Is this a firmware issue?
It never happens when the SD card is inserted before the printer is switched on.

Things I have tried:

- Setting the baudrate to 115200 and 250000 simultaneously in both MKS and Marlin firmware
- Got a new SD card and cardreader
- Formatting with SD Card Formatter
- Rolling back to Marlin-bugfix-1.1.x
- Short file name for gcode.

Problem doesn't occur when printing from a USB stick, but printing from USB occasionally hangs during printing.

Firmware version: Marlin-bugfix-2.0.x and TFT28_32_v3.0.6

mks-viva, 2020-11-30:
Can you provide a video for me when you insert SD card to MKS TFT?

Anima1000000, 2020-11-30:
[IMG_9576.zip](https://github.com/makerbase-mks/MKS-TFT/files/5616015/IMG_9576.zip)

Anima1000000, 2020-11-30:
Thank you for looking into it.

mks-viva, 2020-12-01:
I want to know, what is your SD card format, FAT32 or other?

Anima1000000, 2020-12-01:
FAT32 16GB

Anima1000000, 2020-12-08:
I did a bit of testing. It seems ejecting or inserting the SD card from the main menu reduces the amount of crashes by a lot. So not from within preheat or print. Would a mounting function maybe solve this?