makerbase-mks/MKS-TFT #489: MKS TFT35 V1.0 CNC firmware not communicate with Arduino CNC shield V3

RepRapThailand, 2020-11-30:
Hello,

I can move motor with GRBL sender but not with TFT35.

GRBL Firmware
https://github.com/gnea/grbl/releases/download/v1.1h.20190825/grbl_v1.1h.20190825.hex
******************************************************************************************************

TFT35 CNC
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS_TFT_CNC/MKS_TFT35_CNC_V1.0.2%20Release%20file.zip

[image]

*******************************************************************************************************

Show "connected"  but not connected to the Arduino CNC shield.

[image]

mks-viva, 2020-11-30:
Hi! 
GRBL default baudrate is 115200
Your TFT set baudrate is 115200
So, your firmware settings is ok!
Now, we need check connect of MKS TFT35 to CNC shield V3
MKS TFT35 AAUX-1 need connect to CNC shield V3's TX RX GND 5V
[image]
[image]

RepRapThailand, 2020-11-30:
[image]

[image]

[image]

**************************************************************************************************************************

from TFT35
TX      Brown
RX      RED
GND  Orange
5V      Yellow

mks-viva, 2020-11-30:
You need change connect:
MKS TFT35 TX connect to CNC shield V3's RX
MKS TFT35 RX connect to CNC shield V3's TX

RepRapThailand, 2020-11-30:
ok, success!
Sorry for the stupidity.
ขอบคุณมาก

closed 2020-11-30 by RepRapThailand

Sergii2602, 2022-07-23:
Я так і не зрозумів підключив но мотори не реагують.чи потрібно ще залити прошивку в ардуїно ?

StephenWillis2, 2023-08-22:
Thank you, I needed this info a lot.