makerbase-mks/MKS-TFT #491: MKS TFT35 CNC V1.0.2

nikki-m, 2020-12-05:
Have downloaded MKS TFT35 CNC V1.0.2 to my Makerbase TFT35 board, which I want to configure for CNC use. 
However, there appears to be no z axis control available.
Have I downloaded the latest firmware or have I chosen the wrong files?
Also, what is involved in changing the x and y direction arrows to something more 'sensible' - ie which actually reflect the direction the spindle is moving in. Such as straight up, straight down, left and right. Rather than the diagonal arrows used at the moment.

mks-viva, 2020-12-07:
Hi! Like this, you need copy all to SD card and insert to MKS TFT35 and update it
[image]
[image]
[image]
good luck for you

mks-viva, 2020-12-11:
I will close this issues temporarily

closed 2020-12-11 by mks-viva