makerbase-mks/MKS-TFT #492: no images after setup

fvhohnhorst, 2020-12-13:
i need help with setting up the MKSTFT 2.4 V1.1

when i upload via an SD-Card the firmware, everything is working, except there are only white square where the images should be.
[image]

mks-viva, 2020-12-14:
You need recopy mks_pic and update it
[image]

fvhohnhorst, 2020-12-16:
tried that, but still there are no images shown...

mks-viva, 2020-12-17:
Can you tell me which firmware and version you downloaded?

fvhohnhorst, 2020-12-21:
downloaded the TFT24_V3.0.6 Realase file

DivingDuck, 2020-12-21:
Don't know if you had check this: 
You need to have a complete set of directories and files including mkstft24.bin and mks_config.txt at the same time on the sd-card. The pic-files  will only be uploaded if the config file with correct naming is available and is not called like config.cur / mkstft24.cur. All files have to follow the original naming conventions and locations. No renaming is allowed.

fvhohnhorst, 2020-12-22:
This is on the card before setting it up
[image]

fvhohnhorst, 2020-12-22:
[image]

fvhohnhorst, 2020-12-22:
This is on the sd Card before setting it up

fvhohnhorst, 2020-12-22:
[image]

fvhohnhorst, 2020-12-22:
after testing the fans, all of the images are changed

fvhohnhorst, 2020-12-22:
[image]

fvhohnhorst, 2020-12-22:
this is after setting it up

DivingDuck, 2020-12-22:
Maybe I'm wrong, but shouldn't the file name of mks_config.txt also change to config.cur if it has been successfully processed?

mks-viva, 2020-12-23:
@fvhohnhorst 
I think the motherboard does not actually read the SD card. If it can be read, then all the files will be renamed. I suggest you can try and test it with another card

fvhohnhorst, 2020-12-23:
thanks for your help, with another sd card it finally worked

closed 2020-12-23 by fvhohnhorst