makerbase-mks/MKS-TFT #494: I can move and connected my cnc with GRBL sender or My MKS TFT 35 v1.0 But  I can't send my gcode file to Play Carving

Newbicnc, 2020-12-21:
I have no Problem With Jog on My TFT because my cnc can move smoothly on X,Y or Z axis ...  But when i Start Carving with menu carving on my TFT35  sccreen  My CNC doesn't  want to move or do Anything just stuck ...maybe somebody help me  what its the problem ??  IF I start Carving With GRBL sender In Mylaptop its Ok and No Problem at All .
[image]
[image]

mks-viva, 2020-12-21:
You need download it 
[image]

Newbicnc, 2020-12-21:
That is ..already done.   Sorry maybe this is my fault, because the real problem why my cnc can't move and can't run the carving... That's because the mks Tft 35 v 1 can't read the file gcode in the Folder.. You must  put your gcode file without the folder... And at least my  problem solved . i hope in the future this problem should solved by mks team because this a little bug but.. The people like me that's newbe on  cnc machine very confused and headache. 
By the way  Thank's for your attention.

closed 2020-12-21 by Newbicnc

mks-viva, 2020-12-22:
Thanks for your feedback, we will improve it