makerbase-mks/MKS-TFT #495: Video support for CNC users.

LancingBiker21, 2021-01-02:
Hi Guys.,Could you make an instructional video using a TFT3.5 display and driver board (MKS DLC) running a CNC machine? That is NOT a 3D printer or lazer cutter and using the latest firmware available- just to prove it can be done! 
You advertise on AliExpress and claim that the TFT 3.5 is capable of doing this a lot of buyers would like to see it I'm sure.
Thanks , LB

mks-viva, 2021-01-03:
Hi! You need settings
1. GRBL set CNC mode
2. MKS TFT35 config set CNC mode and firmware use https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS_TFT_CNC/MKS_TFT35_CNC_V1.0.2%20Release%20file.zip
[image]

LancingBiker21, 2021-01-03:
Hi, Thanks for your quick response... I will try to reload the firmware as per your example and get back to you with the results. Would be good if you could provide some screen shoots of the CNC GUI for reference so I know what to expect.

mks-viva, 2021-01-04:
Thank you for your feedback. We will receive suggestions from more users.

closed 2021-01-06 by mks-viva

condigia, 2023-03-04:
> Thank you for your feedback. We will receive suggestions from more users.

However, TFT 35 Having problems reading files .nc . if DLC connect PC run ok. fix it