makerbase-mks/MKS-TFT #496: MKS TFT-Laser

UnleashSpirit, 2021-01-03:
Hi does anyone can use it without any problems ? I can't :(

I always have stupid move across all piece with laser full power which ruin everything !
Can't figure where it comes. I convert SVG to GCode with LaserGRBL.
I tried cuting from LaserGRBL in USB on PC, with CncJs with USB on  via Rpi, directly from SDCard and screen nothing works
I check GCode generated with CAMotics no stupid moves.
 
Any ideas ? I think I'll go back to CNCShield less problems ...

Versions :
boards :
MKS DLC v2.0
MKS TFT35 v1.0
Softwares :
MKS_TFT35_laser_V1.0.3
grbl v1.1h
LaserGRBL v3.8.7

mks-viva, 2021-01-04:
Can you provide some picture about your dlc and tft35 when you use it?

UnleashSpirit, 2021-01-04:
Hi, it's à bit root but here a picture.
I notice that while taking picture, the screen "connected" blink on "disconnected" ultra fast and goes back to connected (with a blue led blinking too on the board)
[image]

mks-viva, 2021-01-04:
I think you wiring is have errors.
[image]
Laser module TTL PWM connect to MKS DLC V2.0 S
Laser module power connect to MKS DLC 12V out
[image]

UnleashSpirit, 2021-01-04:
My LaserPower is on "-Laser+", just below "-12Vo+" and it's working
(picture is bad yeah sorry)
And PWM is wired only on S(ignal). Like the "test board" provided with the DLC

UnleashSpirit, 2021-01-04:
I try again at the moment by changing my power supply (12V 5A) which can be limited to put everything under tension for a more powerful (12V 10A)
No change, I still have ghost movements coming out of nowhere. The screen is marked as "disconnected" when fucking up. I stop cutting with screen, nothing happens for a while then marked as "connected" and stop.

Evgaleks, 2021-01-05:
try to comment this lines in to  mks_config.txt 

[image]

mks-viva, 2021-01-05:
@UnleashSpirit What is the model you actually designed?

warener, 2021-01-05:
If you want to work from the pc via usb you have to disconnect the tft. The other way around, it's the same. if you want to work from the tft, the pc must not be connected. MKS said it is because they both work on the same port.

since I've been following it, it works for me

UnleashSpirit, 2021-01-05:
> If you want to work from the pc via usb you have to disconnect the tft. The other way around, it's the same. if you want to work from the tft, the pc must not be connected. MKS said it is because they both work on the same port.
> 
> since I've been following it, it works for me

Not for me, usb is not connected when using sd card from tft

warener, 2021-01-05:
Used another sd card?

Evgaleks, 2021-01-06:
> If you want to work from the pc via usb you have to disconnect the tft. The other way around, it's the same. if you want to work from the tft, the pc must not be connected. MKS said it is because they both work on the same port.
> 
> since I've been following it, it works for me

My machine works simultaneously and without problems with both PC and TFT35

mks-viva, 2021-01-06:
@Evgaleks Can you share your config or other settings? This will make a lot of sense for other users.

Evgaleks, 2021-01-06:
> @Evgaleks Can you share your config or other settings? This will make a lot of sense for other users.

I use **LaserWeb 4.0.996**  ,  **MKS DLC V2.0** with grbl_1.1h ,  **MKS TFT35 V1.0** with MKS TFT35_LASER_V1.0.3  and **MKS WiFi TFT V1.0** module.

My GRBL config:  [GRBL 1_1h setting.txt](https://github.com/makerbase-mks/MKS-TFT/files/5773794/GRBL.1_1h.setting.txt) _(my machine has an additional Z-axis and End-sensors for x,y,z)_
> $$
 $0=10
 $1=25
 $2=0
 $3=1
 $4=0
 $5=0
 $6=0
 $10=0
 $11=0.020
 $12=0.002
 $13=0
 $20=1
 $21=1
 $22=1
 $23=3
 $24=100.000
 $25=3000.000
 $26=250
 $27=1.000
 $30=255
 $31=0
 $32=1
 $100=80.000
 $101=80.000
 $102=800.000
 $110=4000.000
 $111=4000.000
 $112=600.000
 $120=100.000
 $121=100.000
 $122=100.000
 $130=375.000
 $131=300.000
 $132=45.000

My   [mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/5773824/mks_config.txt)   
At first it was created for MKS TFT35_LASER_V1.0.2 , with four custom battons _( Home , Z +1, Z -1, Z-home)_ but in version 1.0.3 the fourth costom button is not available, 

> #set number of  "More" button 
>moreitem_pic_cnt:4
>
>#edit command for 1~4 "More" button,each command must be separated by semicolon ";"
>
>moreitem_button1_cmd:$h;
>moreitem_button2_cmd:G91; G0 z-1; G90;
>moreitem_button3_cmd:G91; G0 z1; G90;
>moreitem_button4_cmd:G0 z0;

After turn-On the power, execute the "Home" command (use custom buttons - moreitem_button1_cmd:$h;) , after that the machine works normaly, otherwise it freezes.  Also it happens when the g-code coordinates go out of range limts. This is due to an incorrect software connection between GRBL and MKS TFT modules.
I also had to commit the following lines in the configuration file,  otherwise it caused a large number of errors and hangs of the MST TFT35 module.

> #command setting on button1~7
>
>#>morefunc1_cmd:G28;
>#>morefunc2_cmd:G28;
>#>morefunc3_cmd:G28;
>#>morefunc4_cmd:G28;
>#>morefunc5_cmd:G28;
>#>morefunc6_cmd:G28;
>#>morefunc7_cmd:G28;

Display with custom buttons
[image]

 (P.S. files are associated with custom buttons :  _bmp_morefunc1.bin , bmp_morefunc2.bin , bmp_morefunc3.bin , bmp_morefunc4.bin_ )

UnleashSpirit, 2021-01-15:
Hi,
I try laser cutting without screen wired and I don't got any ghost moves ! (laserGrbl via usb)
So it seems that screen as something to do.

I'll try other idea, comment lines in firmware, new SD card etc, I hope this weekend.

@Evgaleks How and where do you wire TFT Wifi module on MKS DLC 2.0 ?

Charus-OV, 2021-01-16:
> Я думаю, что у вас в проводке есть ошибки.
> [image]
> Лазерный модуль TTL PWM подключается к MKS DLC V2. 0 S
> Мощность лазерного модуля подключается к MKS DLC 12V out
> [image]

[image]

UnleashSpirit, 2021-02-04:
Hi,
After more tests everything seems to work fine.
I did all the changes you suggested, comment all the morefuncX_cmd, change the sd card don't work with tft and usb at same time but I kept my wiring as it is.
As I did all together I can't say which one worked but my best guess is the SD card (cause I can't see why comment the morefuncX_cmd would change anything)

Sorry for the issue which isn't one and thanks to all for your help !

closed 2021-02-04 by UnleashSpirit