makerbase-mks/MKS-TFT #499: MKS DLC + TFT35 Laser ON dont work

YuraBz, 2021-01-06:
Hello everybody. Why does not turn on the laser with tft35, at the same time the shutdown works correctly and from  laser grbl also work perfectly.
how and where to fix it. Thank you

Evgaleks, 2021-01-07:
> Hello everybody. Why does not turn on the laser with tft35, at the same time the shutdown works correctly and from laser grbl also work perfectly.
> how and where to fix it. Thank you

Does not turn on from g-code or in manual mode from the display (TFT35), if from the display then check the setting of laser pover from menu.

mks-viva, 2021-01-07:
@YuraBz 
Hi! Can you provide your wiring picture to me?

YuraBz, 2021-01-07:
> @YuraBz
> Hi! Can you provide your wiring picture to me?

[image]
DLS TTL to PS IN & GND

Serial console lasergrbl:
**Laser on from pc**
[22:04:40] [ M3 S10 ] ok
[22:04:50] [ M3 S10 0] ok
**Laser on from TFT35**
[22:05:16] [ ] [HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
[22:05:24] [ ] [HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
**Laser off from TFT35**
[22:06:40] [ M5 ] ok

the connection is correct, the laser works when controlled from the PC, turns off when turned off with tft35 but does not turn on from the stand-alone controller.

mks-viva, 2021-01-09:
According to the picture, there is no problem with the wiring between the motherboard and the Laser module. You need to refer to the laser module manual, such as power supply, mode, etc.

YuraBz, 2021-01-09:
The problem is in the tft35 firmware where I can download the correct one or how can I fix the firmware myself?  thanks

YuraBz, 2021-01-12:
some ideas?

larsvankleef, 2021-01-31:
I have the same problem. I can turn off the laser with the display but putting it on again doesn't work. With other commands the blue led on the makerbase dlc blinks but when I want to control the brightness with the display there is no action.

It works fine when i control the power through my computer.

mks-viva, 2021-02-04:
@YuraBz @larsvankleef 
If you has open Laser opation, MKS DLC motherboard's D1 is will open?

YuraBz, 2021-02-04:
> @YuraBz @larsvankleef
> If you has open Laser opation, MKS DLC motherboard's D1 is will open?

yes

UnleashSpirit, 2021-02-13:
Don't know if it will work for you but to open laser with tft, I had to choose power before.
Set power to 1% in power menu then open/close work (at least for me)