makerbase-mks/MKS-TFT #501: CNC mode actually laser mode

Q5Grafx, 2021-01-13:
I changed the firmware to what i thought was cnc mode but there are laser controls, no spindle controls no z axis control. is there a real cnc interface for this or is it something im going to have to hack and make myself

mks-viva, 2021-01-13:
Hi! Maybe it can't be used like this?

Q5Grafx, 2021-01-13:
i cant set z zero or move z because it has z as a nominal height and there are no buttons for z

Q5Grafx, 2021-01-14:
the cnc firmware that is available does allow me to move my x axis and y axis. there is no z axis manipulation in the interface. i have found the interface tool to make a functional interface tool but the problem is that only seems to work for the 3d printer mode and none of them were able to control my motors. I have 20 years making front ends. if i could be pointed in the direction of where to look for the commands for the cnc version are for direction i could modify and add buttons and make a more workable cnc interface everyone could use. I always give my artwork away freely. Or if i could be told what firmware the cnc version is looking for i could change the code of the 3d printer interface that has the buttons i need for z jog and could then modify the laser settings area to spindle and feed rate settings for a cnc. i really like the display and have used them before making carputers i just need the codes for the buttons so i can make a ui.

levsteka, 2021-01-14:
Hi, I have the same situation. After I changed firmware to (2) CNC there was no Z button. Please provide a configuration file or manual otherwise the device will be useless. Txh Alex

mks-viva, 2021-01-16:
@Q5Grafx @levsteka 
Hi! I has test it and have Z axis in MKS TFT35 display screen
[image]
[image]

levsteka, 2021-01-26:
Thank you for replay. I tried with your FW and it works fine and has Z button. Then I tried to find the differences between your config file and config file which is available online. (FW file is the same.) I noticed some changes and I believe that the name of machine configuration has an impact -  cfg_machine_type vs. cfg_firmware_type. Details are in picture. I suggest more control over development and better testing. Regards Alex
[image]

closed 2021-02-25 by mks-viva