makerbase-mks/MKS-TFT #506: MKS TFT35 Laser - mks_config.txt PROBLEM

Francesco-Rugg, 2021-01-28:
After installing _MKS TFT35 Laser_ and makes change on _mks_config.txt_ (rotate the display of 180 with the formula 180:EE)
once I turn of the tft and power on again I loose all the changes that I made in the .txt file

mks-viva, 2021-01-29:
Hi! Can you provide your config to me and i will test it.

Francesco-Rugg, 2021-01-29:
[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/5892933/mks_config.txt)

mks-viva, 2021-02-01:
It seems that everything is normal. After you update, has the name of mks_config.txt changed to ***.cur?

Francesco-Rugg, 2021-02-03:
> It seems that everything is normal. After you update, has the name of mks_config.txt changed to ***.cur?

yes

mks-viva, 2021-02-04:
If has change name to ***.cur, the config has loaded.

Francesco-Rugg, 2021-02-04:
> If has change name to ***.cur, the config has loaded.

Ok but the display when I turn on again the TFT screen turn to 0 position, not 180°

One more question.
In the config file where I can set the X-Y maximum travel?

mks-viva, 2021-02-06:
@Francesco-Rugg 
X-Y maximum travel set, you need you “$130” and "$131" command, you can refer to here:https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

Francesco-Rugg, 2021-02-06:
> @Francesco-Rugg
> X-Y maximum travel set, you need you “$130” and "$131" command, you can refer to here:https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration
> [image]

I need to set up the TFT, not the DLC 2.0.

Francesco-Rugg, 2021-02-06:
> Ok but the display when I turn on again the TFT screen turn to 0 position, not 180°

This problem is not resolved.

mks-viva, 2021-02-06:
Provide you config file and firmware to me and i will test it.

Francesco-Rugg, 2021-02-06:
> Provide you config file and firmware to me and i will test it.

@mks-viva 
8 days ago you ask me for the CONFIG file and told me that the file is OK, but still have problem

this is the file

> [mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/5892933/mks_config.txt)

m-sja, 2021-02-25:
I can confirm the same behavior on my unit with this firmware (English Folder):
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT-laser/MKS_TFT35_laser_V1.0.3%20Release%20file.rar
Only change in mks_config.txt is the Line for screen rotation to:
`>cfg_screen_overturn_180:0xEE`

After inserting the sd-card and first powering the unit, the screen is rotated by 180° as intended.
(The green Bootloader-Message "Booting ..." is rotated and all the Interface is also)
The File on sd-card changes to ***.cur

After first reset or Power-Cycle, the green bootloader-message "Booting ..." is still rotated, but the whole interface is changed back to default orientation. This is not changed on further reboots.

mks-viva, 2021-02-25:
@m-sja @Francesco-Rugg 
Hi! I has test it, this is a bug, i will feedback to our engineer

m-sja, 2021-03-20:
Hi @mks-viva 
is there any timeline when we could expect the fix?
It still looks wierd if the Interface is upside down.
[image]

Best regards

danielminson, 2021-03-28:
I also am having exactly the same problem. Please fix it ASAP
[image]

danielminson, 2021-04-24:
> Hi @mks-viva
> is there any timeline when we could expect the fix?
> It still looks wierd if the Interface is upside down.
> [image]
> 
> Best regards

I am really need this issue to be fixed? Is there any update on it?

Francesco-Rugg, 2021-04-24:
@danielminson I haven't tried yet, the problem is the same of 3 months ago?

danielminson, 2021-04-24:
Yes, the latest version of the firmware TFT35laser is 1.0.3 and it has exactly the same issue described here. It´s such a bummer. Is there any way to tag Makerbase to actually fix the issue?

danielminson, 2021-06-17:
@makerbase-mks @Francesco-Rugg still waiting for a fix.