makerbase-mks/MKS-TFT #508: TFT load/unload not working Marlin 2.0.6

wildvortex, 2021-01-30:
I have been working on this build. First I was able to unload with the touch screen but not load. Had a bad driver so replaced and had to change the direction. At the same time I changed max feed to 1100 (bowden length plus 100) . I loaded the bin file and great all of a sudden it's loading. But, when I went to run unload now it is not working. 

I will note that the screen runs like it is working even showing the bar moving but, the stepper does absolutely nothing.

mks-viva, 2021-02-01:
Hi! Which version of TFT firmware did you use? Because TFT only sends instructions, the entire execution is still in the marlin firmware

wildvortex, 2021-02-01:
Can close this issue. I believe it was due to unload speed. Can't say for sure but it is now working. Thank you

closed 2021-02-01 by wildvortex