makerbase-mks/MKS-TFT #510: MKS TFT35 V1.0

angeldrug, 2021-02-05:
Hello. I bought TFT 35 v1.0. Several questions arose.

1.What needs to be turned on for TFT 35 v1.0 to see the motherboard sd card?
SDSUPPORT - enabled and the old display sees the SD card, but the TFT 35 v1.0 does not.

2. I set up a custom menu and the 7th button does not work - there is an icon, but it does not execute the command.
I move it to button 6 and write to config.txt -
works, I return it back - does not work.
7 buttons are included in config.txt, but the 7th button does not want to work.

3.And it is not clear how Z-offset works? I press to raise the Z-axis by 1mm, the printer asks for confirmation
- I confirm and nothing happens. Z-axis does not rise. This function is TFT at all
It works, thanks.

mks-viva, 2021-02-06:
Hi! About your issues:
1. MKS TFT have not support see the motherboard sd or tf card function. I am sorry; 
2. Provide your config file to me and i will check it
3. Which interface are you operating on?

angeldrug, 2021-02-06:
Thanks for the answer. Marlin 2.0.6.1.
[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/5936335/mks_config.txt)

And2029, 2021-02-12:
Здравствуйте.Когда появится ось Z в прошивке CNC на экран TFT 35 v1.0 ????