makerbase-mks/MKS-TFT #511: TFT35 with laser firmware

AleBS71, 2021-02-05:
Hello, I have e problem with my new configuration with MKS- DLC and MKS- TFT35. All work fine when i try to engrave a file, the problem is only at the start when I send the command (from the touchscreen) to show me the edges of the file to be processed, the displacement size of the machine is at least double the work to be done. If I then start the process, it does it correctly. Can you help me for that ?
[image]

mks-viva, 2021-02-07:
Can you provide your config and bodies when laser work?

tpaolillo, 2021-03-01:
Hi, What firmware are you using for your TFT35? Where can i download it? Thanks

ElectricalUK, 2025-12-16:
I have the same problem in 2025. 

1) The command BORDERS horizontally runs twice too far, but vertically seems fine.
2) Also  Borders / Figures / 'Cancle'   - supposed to be CANCEL

P.S 
Using a CNC configuration for the laser because the laser configuration doesn't have a Z axis.

MKS_TFT35 1.0.3 CNC