makerbase-mks/MKS-TFT #512: **MKS_TFT35_CNC_V1.0.3 Release file-  Not Useable **

sivaelid, 2021-02-05:
**Issues with the TFT35_CNC   V1.0.3. HAs this been though QA as this is very bad  who checked IT? and if MKS release this software how bad is the rest?

 ISSUES**

1.	Can not select Lazer / CNC  i.e no Z  up down buttons for CNC 
2.	Multi Language can not be turn on and off with Config file setting
3.	Change of colours of buttons et in the config file do not change the display
4.	 The move buttons are now missing the + And – symbols  and No Z Button for the CNC mode
5.	The Movement increment arrow  that showed 0.1,1,10 mm now does not so can not adjust
6.	The Version   shows in red Version 1.0.3 But when selected for CNC does not show this or alter the buttons
7.	More buttons seem not to work
8.	Speed setting Arrow now has no increments shown though the +/_ button alters in different increments
9.	Speed button does not seem to work being on/off
**While its nice to see WiFi being added the items above are  major issues and a make the unit unusable**

levsteka, 2021-02-09:
I tried the new firmware and my conclusions are:
- there is still an error in first line cfg_machine_type vs. cfg_firmware_type. If I change to cfg_machine_type:2 then firmware actually work as a CNC firmware;
- unfortunately, the device still doesn't work as it should. When I boot CNC TFT successfully connect (status connected) but when I touch any button on TFT 35 device disconnected immediately. I checked the status on PC and if I reset ($X) then device reconnects. There are obviously quite a few problems.  I would be very happy if you eliminate them. Regards Alex

m-sja, 2021-02-25:
The problem is, that the TFT sends commands to GRBL every few seconds to ensure proper connection.
But in Locked-State, GRBL is not responding in a way the TFT can handle and it shows "disconnected".
The bad thing is, that all commands you send via touch-buttons are buffered in the TFT until a connection is established with GRBL (in a way the TFT expects it).

So Custom Buttons with $H or $X are not working while GRBL is in Locked/Alarm State...

You can only make a custom Button with $X or $H and this is the first button you must press after powering everything on.
It works until GRBL locks due to any reason (Soft/Hard-Limits or any other error...). After that you must powercycle TFT and GRBL wich makes it totally useless as an standalone offline controller.

francois14-decroix, 2021-02-28:
how do we add a button with the $ X and $ H functions because I have the same problem with a laser version
I tried the MKStool software I can't see the icons for the laser or cnc

m-sja, 2021-02-28:
@francois14-decroix
in the root of your sd-card you find the `CONFIG.CUR` file.
Rename it to `mks_config.txt`
Than edit these Lines:
```
>moreitem_pic_cnt:2
>moreitem_button1_cmd:$H;
>moreitem_button2_cmd:$X;
```
Save the file, insert the sd-card and power on the display.

Due to the current issues with the firmware, it is not possible to recover connection to GRBL once GRBL enters alarm-state.
It is also not possible to create more than 3 user-buttons.
The very first Button you have to press is $H or $X (depending on your machine and configuration).
See Issue #431 on that.

francois14-decroix, 2021-02-28:
Thank you for your reply

I managed to add 2 buttons
how can we modify the appearance of these buttons because with MKStools we open buttons for a printer but not cnc and laser
in the drop-down menu we have Mks robin tft 3.5 and do not see any laser or cnc icon

fandres, 2021-05-03:
Unfortunately the cnc / laser firmware is very green

How can I compile the firmware for the CNC? or where can I get the source code of TFT35 CNC?

For those who require the size of the icons to be modified are the following:
Logo: 480x320.
Icons: 117x140.
Use image2lcd.

Documentation does not exist

bnvelozo, 2021-05-06:
I've been trying to add those button withou success. There is any step to do before edit those lines? Can anyone please send the .txt file? thanks