makerbase-mks/MKS-TFT #513: CNC Firmware issues and need for open source TFT Displays

sivaelid, 2021-02-05:
Hi the MKS screen is very good but because it is  not open source the issues and mistake do not get fixed  Either can we have working version of the CNC firmware  not Just working for lazer: as at present   firmware has no Z axis control.   Why not  post in github the source code so the community can  amend and update . 

You will sell more product if the CNC function is working well as there are few options for Touch screens for  CNC / Lazer control for machine like the CNC3018 or4030 etc.

m-sja, 2021-02-25:
+1 on this one. Please open-Source the Laser/CNC-Firmware as well.

I just received the MKS 35 TFT and it is a nice piece of hardware, but totally useless for CNC or Laser.
- Disconnected when GRBL is in Locked State, no way of sending $H or $X, cause the commands are Buffered until GRBL is in Idle state
- No M4 Support (Relative Power in GRBL Lasermode)
- "Open/Close" Button is misleading and dangerous cause it enables the Laser without movement which will cause fire
- Button-labels are misleading. I.e. "Home" is not Homing, but "Return to Work Zero". "Open/Close" is Laser "Enable/Disable", ...
- Only 3 custom buttons possible (But in Config File there are more options to set up to 4 + 7 Buttons)
- Screen rotation feature is broken, it just works only on first start after uploading new firmware, but after reboot only bootloader is rotated, the interface is not rotated anymore.
- Instead of "Connected/Disconnected" it would be more useful displaying the GRBL state "Idle,Busy,Locked, ...
- No error codes returned, it just shows "Disconnected". Schwing the Error-Number and have a Button to unlock GRBL is essential.

Most of it could be fixed easily if the source code would be available.
You already released the 3D-Printer source code, please do it as well on CNC/Laser

Thanks and best regards

Clubheader, 2021-03-04:
Also, on the firmware for a laser or cnc, the function of continuing printing when the power is turned on does not work. The display just goes into reboot when I hit continue. This is very sad, because it was because of this function that I took the MKS TFT.

Sprint8, 2021-03-20:
I concur that the Z axis buttons are gone and I've seen them in some videos of the TFT35 controller which probably has an older FW.

m-sja, 2021-03-20:
@Sprint8 
sadly the Firmware is not usable for GRBL.
There are to many issues.

However for your problem there is already a known fix.
Pleads see the first point on this post.
https://github.com/makerbase-mks/MKS-TFT/issues/512#issuecomment-775744997

I hope this heps you get one step further.
Please post and comment any further issues you encounter.
Sadly we need some pressure on MKS as Laser/CNC seems not getting attention it deserves.

@makerbase-mks  @mks-viva 
This is the only GRBL compatible “nice“ offline Controller on the Market (correct me if i missed one).
It is so sad that it is not programmed well and issueds are not solved. Due to that it is not usable.
Please give better/faster support or publish the Source.
The Community offers so much skilled programmers and CNC/Laser experts who are willing to improve your product for free!

Best regards

Sprint8, 2021-03-20:
@m-sja I totally agree that this is probably the only nice grbl controller. BigTreeTech has some decent touchscreen too and their firmware seems to have a good user following, but no Inc version.  I wonder if they'd be willing to port their code to a ctc version? It'd be a simple task for them to pare down the 3d printer code vs someone like me hacking it. (Haven't touched coding since college).

And thanks for the button tip. It's a workaround, I'd rather have a vendor supplied solution for a feature that should be there in the first place.  
I totally agree with everything you said.  The user community would definitely participate to enhance their brand

Sprint8, 2021-03-20:
I've put in a feature request to BTT's TFT35 github for the creation of CNC/laser FW. I'll let you know if I get a response and what came of it.