makerbase-mks/MKS-TFT #515: Babystep stop printing

erdemarslan, 2021-02-09:
When i use babystep on TFT35 (1.0.8) with Marlin 2.0.7.2, it stops printing.

Firmware: Marlin 2.0.7.2
TFT35 Firmware: 1.0.8
Slicer: Cura 4.8.0

How i can solve this?

closed 2021-02-11 by erdemarslan

Magiquall, 2021-03-04:
Hello erdemarslan,
I have the same problem, have you found a solution?
Thank you

erdemarslan, 2021-03-04:
> 
> 
> Hello erdemarslan,
> I have the same problem, have you found a solution?
> Thank you

Hello Magiquall.
I regret using MKS-TFT. Instead I bought BTT-TFT35 and I'm using it. My solution is this. Unfortunately, MKS does not stand behind its products. Updates are coming late, not all problems are solved in a timely manner. I no longer use MKS products.

Magiquall, 2021-03-04:
Thanks for your quick comments,

I actually have the impression that MKS is abandoning these customers, I have contacted them several times without any feedback.
But a friend, has an SKR and is also in trouble. But I feel like there are more people in the community using SKRs, On my ender 3 I have an SKR Mini E3 and it works perfectly.
And I started with an MKS SGEN_L V2.0 for my Hypercube, but not many people bought this board.

Thank you in any case good continuation.

erdemarslan, 2021-03-04:
One upon time i bought mks sgen l v1 and i cant upload marlin because there was no marlin support. Then i bought skr 1.3 and still i use it. After I didnt change it. Skr is powerful for me. Now i use btt tft35 and its better then mks tft35. I will never use mks products after this time