makerbase-mks/MKS-TFT #518: Loosing connection MKS DLC Dead ?

UnleashSpirit, 2021-02-13:
Hi,

After lots of trouble, i finally make it works... for a while
Now I'm loosing connection to MKS. Etheir with TFT35 or LaserGRBL usb (never both at the same time), after one or two moving command, connection is lost and I can't do anything, pause, stop close laser, nothing.

I already try to flash board with Xloader or Arduino IDE (arduino nano) nothing to do. Reinstall TFT Laser 1.0.0 neither.
Is my board dead, a new one can work better ?
[image]

mks-viva, 2021-02-14:
If you only use DLC motherboard connect to pc by USB and it is ok？

UnleashSpirit, 2021-02-14:
No 
Neither USB without tft not tft without usb work, I always lost connection

UnleashSpirit, 2021-02-14:
Here my last try (same result via ubs or sdcard+tft)
[image]

From this gcode :
[dessus_3mm.txt](https://github.com/makerbase-mks/MKS-TFT/files/5977962/dessus_3mm.txt)

I try to pause/stop from TFT no response ("disconnected")

mks-viva, 2021-02-14:
Grbl default bardrare is 115200

UnleashSpirit, 2021-02-14:
That's the baudrate i'm using

burneme, 2021-04-11:
I had similar problems with communication between TFT35_laser Ver.1.0.3 on MKS TFT35 V1.0 and a two axis Arduino Uno board based board. I could manage to define extra buttons to send a "$X" to unlock and "$H" to home. Then it works.
Unfortunately control boards share the single serial port with the CH340 circuit (for the USB-input) and the so called "remote connection" (AUX). Here they use resistors in the RX and TX-lines to get it feasable to cooperate. Furthermore they add LEDs for signaling directly into this resistor network, therefore the low level for the receive pin of the ATMEGA 328P is too high. I removed the LEDs and reduced the resistances between TXD of MKS-TFT35 to control board. Then it is permanently (green) "connected".

But I still have communication problems in a longer job coming from the SD-Card. It works fine for a while, but when i.e. laser power is changed (new layer) it misses steps and burns into the wildness (but reproduceable), like shown in Unleashed_Spirits's picture. Now I plan to burn directly from Lightburn thru the USB-port to see whether the problem still exists. Will report soon.

Is the MKS-TFT35 firmware open to see?  Is the X-on/X-off protocol used? I suppose it has something to do with buffer-overrun.

burneme, 2021-04-12:
Sending directly from Lightburn-> USB (comport) works fine.
Will log the data sent from MKS TFT35 when printing from SD-Card, compare and report here.

burneme, 2021-04-14:
[image]
The low-level of the TXD (from TFT35) ist still bad because of the load of the CH340 chip, even it is not connected to the pc.
I have to think about an intelligent workaround to avoid this problem. Will come back.

burneme, 2021-05-04:
The easiest way to hold permanently connection from TFT35 or the USB (CH340) was just to set an small switch where I can select either the TX pin of the TFT35 or the TX-Pin of the CH340 towards the RX pin of the atmega328P. Now it is working from Computer or SD card via TFT35.
[image]
[image]

Dimaofficial, 2024-12-13:
[image]
[image]
[image]
[image]
[image]

Пишит disconnected 
Как исправить, прошивку через  SD карту обновобновил но не работает, что дальше делать?