makerbase-mks/MKS-TFT #521: add button

Yetreyert, 2021-02-23:
is it possible to add the auto level button to the manual level screen? has 2 free spaces to use

mks-viva, 2021-02-24:
Hello! Usually only one leveling method is used, and the user does the same. Of course, we will collect more suggestions.

Yetreyert, 2021-02-24:
It depends. If I change the plate you must first align it well and then I create the mesh or start with self-leveling of the bltouch. I think it is essential as a thing to not always depend on the PC. It is a request that was also made by the entire Italian community. I'm trying to understand how the various pages have been set up also to add an on and off lights command

mks-viva, 2021-02-24:
Use MKS TFT35?

Yetreyert, 2021-02-24:
Yes v2 with mks robin nano 2.0. Unfortunately I also saw that there is a problem with the neopixel library too bad

Yetreyert, 2021-02-24:
Yes v2 with mks robin nano 2.0. Unfortunately I also saw that there is a problem with the neopixel library too bad
--
Inviato da Libero Mail per Android mercoledì,