makerbase-mks/MKS-TFT #522: MKS TFT 35 v1.0 on SGEN-L v1.0

organizedkaos, 2021-02-24:
I purchased [this MKS TFT 35 v1.0](https://www.amazon.com/gp/product/B07WGNZJKZ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) to connect to my SGEN-L v1.0.

I have been using my SGEN-L with no issues with the Reprap Discount Full Graphics Controller.

I will have the item soon but I was trying to pre-configure marlin 2.0.7.x for it.

I have gotten some tft use from the mks robin nano v1.2 and tft 35 combo and I prefer the "TFT_COLOR_UI" option in marlin.  To make that work on the robin setup I did #define MKS_ROBIN_TFT35 and then #define TFT_COLOR_UI.

It compiled and worked.

I am having trouble finding how to do the same for the MKS TFT 35 v1.0 on SGEN-L v1.0.  So far it appears that MKS TFT 35 is completely driven by it's own firmware.  I just want the MKS TFT35 to be a slave to marlin and not the master (if that makes sense).

mks-viva, 2021-02-24:
Hello! I think you may have bought the wrong touch screen.
MKS SGEN_L + MKS TS35 V2.0 :Support Marlin color UI, like this: 
https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/MKS_TS35

mks-viva, 2021-02-24:
If you use MKS TFT35 on SGEN_L, you can refer to here: https://github.com/makerbase-mks/MKS-SGEN_L-V2/wiki/Display_MKS_TFT35

closed 2021-02-24 by mks-viva

organizedkaos, 2021-02-24:
Well that's just great.  According to the link:
**When testing or using MKS TFT35, MKS SGEN_L should not be connected with USB cable because of serial port conflict**

Apparently I can't connect my SGEN-L board to my raspberry pi via usb when the MKSTFT35 is connected.

Rather than buy the TS35 I'm just gonna return the TFT35 and stick to the Reprap.  I don't need all that fancy looking buttons since I can run that gcode myself from the console of my repetier server.  What I do need is basic display information (that the TFT35 apparently doesn't offer) and on very small occasions the need to adjust some settings on the fly.

I will be looking at alternative motherboards from now on.  MKS seems to break their own ecosystem with these hardware versions and no backward compatability.

The SGEN-L is capable of exposing the SDCARD on the board to the controlling computer system via USB but the MKS Robin Nano V1.2 is incapable of the same ability (as an example) yet the robin nano tft35 can be switched to marlins color ui from marlin itself.  

Farewell.

mks-viva, 2021-02-24:
Haha，MKS Robin Nano V3（STM32F407VGT6） + MKS TS35 V2
1. Support u disk
2. Support read motherboard’s card and disk via USB
3. Support MKS UI, Marlin Color UI, Marlin 12864UI
4. TMC UART mode... 
More information，You can refer to MKS Robin Nano V3 project on github

organizedkaos, 2021-02-24:
> Haha，MKS Robin Nano V3（STM32F407VGT6） + MKS TS35 V2
> 
> 1. Support u disk
> 2. Support read motherboard’s card and disk via USB
> 3. Support MKS UI, Marlin Color UI, Marlin 12864UI
> 4. TMC UART mode...
>    More information，You can refer to MKS Robin Nano V3 project on github

Which requires the purchase of entirely new hardware.  There is nothing wrong with the current boards I am using and I need no further “enhancements”.  I simply wanted to add/upgrade the screen.  But even adding the mks hardware to mks hardware will break it by the wiki information.  This appears to be the reason there is no backward compatibility.  “If you want this you have to buy all this extra hardware even though your existing hardware is working with no issues”.   If I am going to buy new hardware it will be another manufacturer/company.