makerbase-mks/MKS-TFT #523: MKS TFT35 CNC - SOLUTION DISCONNECTING WHEN PRESSING ANY BUTTON

Sferatos12, 2021-02-24:
Hello, i think i found a temporary solution for the disconnecting problem  when pressing any button.
Grbl 1.1 has a safety feature that not allows you to move or do anything before you home you're machine.
So when pressing anything on the LCD resulting in an error, making the MKS35 disconnecting.
The solution for that, temporary, is to disable the safety feature in the grbl.
You can disable it in config.h with commenting HOMING_INIT_LOCK .
Bu it would be nice to have a UNLOCK function on the main screen of the LCD (X$)

mks-viva, 2021-02-24:
Hello! You can set it by mks_config.txt file, like this
```
#how many "More" button display on screen interface
>morefunc_cnt:1

#command setting on button1~7
>morefunc1_cmd:$X;
```

Sferatos12, 2021-02-24:
> Hello! You can set it by mks_config.txt file, like this
> 
> ```
> #how many "More" button display on screen interface
> >morefunc_cnt:1
> 
> #command setting on button1~7
> >morefunc1_cmd:$X;
> ```
it doesn't show up on the screen....

mks-viva, 2021-02-24:
> > Hello! You can set it by mks_config.txt file, like this
> > ```
> > #how many "More" button display on screen interface
> > >morefunc_cnt:1
> > 
> > #command setting on button1~7
> > >morefunc1_cmd:$X;
> > ```
> 
> it doesn't show up on the screen....

#set number of  "More" button 
>moreitem_pic_cnt:1

Sferatos12, 2021-02-24:
> > > Hello! You can set it by mks_config.txt file, like this
> > > ```
> > > #how many "More" button display on screen interface
> > > >morefunc_cnt:1
> > > 
> > > #command setting on button1~7
> > > >morefunc1_cmd:$X;
> > > ```
> > 
> > 
> > it doesn't show up on the screen....
> 
> #set number of "More" button
> 
> > moreitem_pic_cnt:1

works now. but i think this should be on the main screen, as a default button, in my opinion.
Thanks for hepling.

mks-viva, 2021-02-24:
Okay, thank you for your suggestions, we will collect more user suggestions

closed 2021-02-24 by mks-viva

erikprosa, 2021-06-24:
My TFT35 also gets the disconnected message constantly, when changing the mks_config.txt it doesn't show me buttons. Only if I change moreitem_pic_cnt, and still to no avail.

Sferatos12, 2021-06-24:
I resolved it by adding a custom button with the unlock function

erikprosa, 2021-06-24:
Most forums say that to create the buttons I must change mks_config.txt
Specifically in “#how many "More" button display on screen interface”
But it didn't work.

```
#how many "More" button display on screen interface
>morefunc_cnt:1

#command setting on button1~7
>morefunc1_cmd:$X;
```

Changing “#set number of "More" button", I managed to add buttons

```
#set number of  "More" button 
>moreitem_pic_cnt:1

#edit command for 1~4 "More" button,each command must be separated by semicolon ";"
>moreitem_button1_cmd:$X;
```

However, when the screen is indicated on the display as disconnected, the buttons are useless.

I have to keep turning my machine on and off, taking and putting the usb cable, until the screen connects.

As I'm using the MKS DLC v2 board with the MKS TFT35 on an 80W Co² laser machine, I'm thinking it might be some variation in my electrical network. Because when you turn on the machine, equipment such as TV and monitors turn off and on right away.
I purchased a 3kva stabilizer to see if it improves anything.

Sferatos12, 2021-06-24:
Do not use usb connection when using the dlc screen.

erikprosa, 2021-06-24:
Thanks, I will test it.
If only I could load web view to the MKS Wifi it would help, but it only gives the error, regrettable.

[image]

Sferatos12, 2021-06-24:
I didn't try the WiFi function of the interface. Upload it via sd card

erikprosa, 2021-06-24:
I removed the USB cable, tried to run a gcode straight from SD and it still persists in disconnecting.

https://user-images.githubusercontent.com/64697394/123336377-29adf900-d51c-11eb-91c8-db08a8687334.mp4

Sferatos12, 2021-06-25:
I will send you my config files to check if it works.

erikprosa, 2021-07-05:
Awaiting Despatch.
Even with the USB cable disconnected, the TFT35 is still disconnected constantly.

Sferatos12, 2021-07-05:
Sorry mate! Been busy lately and forgot totally about sending the file. I
will get a reminder not to forget :)

nfroadin, 2022-07-19:
Hi @Sferatos12 
I have the same problem, can you send me also your config file to compare?

1957157, 2023-11-10:
> Привет @Sferatos12 У меня такая же проблема, не могли бы вы прислать мне также свой конфигурационный файл для сравнения?

У меня такая же проблема, все описанное я проделал, результат ноль. Если возможно пришлите рабочий конфигурационный файл.