makerbase-mks/MKS-TFT #533: MKS TFT24 CNC milling from SDcard error

LaGuiche25, 2021-04-16:
Hello,

I have a data transfer problem when I mill a file from the TFT display's SD card.
When I mill with this file from USB everything is OK.

grbl 1.1h
MKS TFT24 display
Firmware MKS-TFT24_CNC_V1.0.3.2

Here is the gcode of the file:
[grbl_code.txt](https://github.com/makerbase-mks/MKS-TFT/files/6325912/grbl_code.txt)

and this is what grbl receives (echoed by grbl)
[grbl_echo.txt](https://github.com/makerbase-mks/MKS-TFT/files/6325908/grbl_echo.txt)

What I have tried to fix this problem:
- Grbl version change -> no change
- TFT firmware version change -> no change
- TFT screen change -> no change
- SD card change -> no change
- SD card formatting -> no change
- Baudrate change -> the error occurs on another line

Has someone already had this problem?

condigia, 2023-03-04:
I also have the same situation

condigia, 2023-03-04:
> 

How Change .txt to .gcode ??? thanks