makerbase-mks/MKS-TFT #551: MKS TFT35 LASER UI

EBurgan, 2021-08-19:
I'm having an issue with the UI on my MKS TFT35 board. I recently bought it to upgrade an existing homemade laser. The issue is that no matter firmware I upload he only UI that loads in is for a 3D printer. I cant get the UI for a laser engraver to pop up let alone open it. I'm probably missing something super obvious

mks-viva, 2021-08-19:
You need download it and update. https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT-laser/MKS_TFT35_laser_V1.0.3%20Release%20file.rar
[image]

EBurgan, 2021-08-19:
Thank you! I could have sworn i had already done that!

EBurgan, 2021-08-21:
So I have it update but no matter what I do or change the axis wont move the correct amount. I tell it to move 10mm and it moves 2mm. Having the same issue with both the X and Y axis. I have tried to adjust the voltage to no change.

nfroadin, 2022-07-19:
Are the Grbl-settings Steps/mm for all axis correct? Inserting the correct Values may be the solution ;)