makerbase-mks/MKS-TFT #577: tft 35 mks genl 2.1 Not start printing

Deluxor270, 2024-02-13:
I am using mks tft35 v1 and mks genl v2.1 product with marlin software. Even though I think I did everything right, I still can't print from the SD card or USB. When I enter the print screen, the temperature values freeze. When I cancel the print on the screen, everything goes back to normal. I have not been able to solve this problem for 3 weeks. I could not find an answer to this problem in the forums. please help. thanks..

Deluxor270, 2024-02-14:
@makerbase-mks ?

Deluxor270, 2024-02-15:
and I noticed something.  If I connect the motherboard to both the display and the computer at the same time, I can print without any problems.  I can't figure out what causes this...

staff1010, 2024-08-23:
If you are using cura slicing, try deleting the M105 instruction in the gcode file.

ahtf, 2026-08-24:
I have the same problem and seeking the answer